SmartEVSE/SmartEVSE-3 #104: MQTT ChargeCurrentOverride not setting any value.

netpet, 2025-08-02:
**Describe the bug**
I am working with a SmartEVSE FW 3.9.1 (3.8.5 has the same problem) standalone, one SmartEVSE, no sensorbox, no Meter, in NORMAL Mode. 

I want to change the Charge Current to a specfied value between 6 and 16A with MQTT. Doesnt work. 

**Upload your config**
uploaded
[settings.json](https://github.com/user-attachments/files/21558547/settings.json)

**Upload a debug log**
JUSTBEFORE.log before the MQTT Publish started
[JUSTBEFORE.log](https://github.com/user-attachments/files/21558546/JUSTBEFORE.log)
JUSTAFTER.log a few hundred milliseconds before the MQTT Publish started until a few seconds after. 
[JUSTAFTER.log](https://github.com/user-attachments/files/21558545/JUSTAFTER.log)
(Actually there is no sign of any information in the debug.log, concerning the problem)

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in?
NORMAL
3. What steps do you take so the problem arises?

MQTT data of the broker was put in via the WebInterface

Mosquitto Broker is working on the RPi.

Subscription works (Output):

SmartEVSE-xxxx/connected online
SmartEVSE-xxxx/Mode Normal
SmartEVSE-xxxx/MaxCurrent 130
SmartEVSE-xxxx/ChargeCurrent 130
SmartEVSE-xxxx/ChargeCurrentOverride 130
SmartEVSE-xxxx/Access Allow
SmartEVSE-xxxx/RFID Not Installed
SmartEVSE-xxxx/State Charging
SmartEVSE-xxxx/Error None
SmartEVSE-xxxx/EVPlugState Connected
SmartEVSE-xxxx/WiFiSSID INNVILLAGE_mobil
SmartEVSE-xxxx/WiFiBSSID D8:32:14:7D:00:21
SmartEVSE-xxxx/OCPP Disabled
SmartEVSE-xxxx/LEDColorOff 0,0,0
SmartEVSE-xxxx/LEDColorNormal 0,255,0
SmartEVSE-xxxx/LEDColorSmart 0,255,0
SmartEVSE-xxxx/LEDColorSolar 255,170,0
SmartEVSE-xxxx/LEDColorCustom 0,0,255
SmartEVSE-xxxx/NrOfPhases 3
SmartEVSE-xxxx/ESPUptime 1875
SmartEVSE-xxxx/WiFiRSSI -70
SmartEVSE-xxxx/ESPTemp 29

SMARTEVSE is posting data, also updating it when changing the Override via the Webinterface, although the posted topic is not "CurrentOverride" as in the documentation but "ChargeCurrentOverride". 

Changing the Override via the Webinterface works

Changing the ChargeCurrentOverride via MQTT on the Rpi doesnt work:

mosquitto_pub -h IP_MQTT_Broker -u USER -P PASSWORD -t 'SmartEVSE-xxxx/Set/ChargeCurrentOverride' -m 90

gets published, but has no effect, also not when changing " instead of ' or putting the 90 in "90" or '90', also not using "CurrentOverride" instead of "ChargeCurrentOverride" and of course with the right credentials for the IP, USER and Password. 

Message gets posted on the Broker, any subscriber could see it, bot no change of the ChargeCurrentOverride at the SmartEVSE

**Expected behavior**
change the Charge Current on the SmartEVSE to a specfied value between 6 and 16A or turn off charging completely when sending less than 6 with MQTT when I publish a Command. 

**Screenshots**
-

netpet, 2025-08-02:
As mentioned also tried CurrentOverride but has no effect.  SmartEVSE is not publishing CurrentOverride, it publishes ChargeCurrentOverride. But either way no change of the override value. May you have an older SmartEVSE device ? Mine was delivered 1 week ago. 

Sending a http request with curl on the pi (same command as the web interface sends) works fine. 

Are there any other topic I could try to change, to see if it only the Current Override or all changes denied ?

netpet, 2025-08-02:
Shame on me, now it works. Maybe you did some magic ?

Another Thing I saw when charging my Tesla is:

When I stop charging on the car side, charging will be interrupted, SmartEVSE states ready for charging but the tesla says "AC Charging not possible. The system will retry shortly". And this goes on and on. After some time Tesla goes in Fault mode and you need to unplug and plugin again. 

Same failure when using "Pause" from the webinterface but not when using "OFF" from the webinterface.

netpet, 2025-08-04:
Contactor is opening (clicking is just fine, or do you mean something else), Tesla is also restarting (also with a clicking of the contactor beside), if it takes not to long for the restart. Problem occurs especially overnight, so when there is a longer time without restart, then you need to unplug, because Tesla states a faulty connection and doesnt want to reconnect.

On the picture you see, how it looks like, when stopped after a few second. Although I have stopped the carside, Tesla still communicates with the "Charger" (obviously because of the still plugged cable).

This doesnt show up, when stopping charging with a  conneced Charger Cable like the BMW Flexible Fast Charger or the well known Juice Booster. They just cancel and wait for reactivating, no comments from the Teslaapp

[image]

dingo35, 2025-08-04:
It is scarce on CP info, but you can get a debug log if you telnet to your SmartEVSE if you flashed a debug version.

dingo35, 2025-08-06:
Stay away from that stuff, is intended to activate a connected HomePlug modem.

You can put _LOG_A statements in SetCPDuty to see what DutyCycle is sent to the CP pin, and in Pilot() you can log the returned pilot variable.
If you want it reported repeatedly you can put log statements in Timer1S_singlerun.

dingo35, 2025-08-07:
Yes I thought you were going to rebuild the firmware since you were on the verge of manipulating the PWM signal on the CP pin.

Ok so I understand it all worked, then you started fiddling around with MQTT and then it stopped working, right?

Did you try testing withMQTT disabled?

Did you check the PE connection of the SmartEVSE, if that is not wired according to specs you can get these kind of problems..

If all fails, do an /erasesettings and try again!

dingo35, 2025-08-07:
By the way:
"PE to PP smartevse (with resistor in between, 220 Ohm)"

This might be your problem, if you are using a fixed cable the resistor is already in your CCS plug!

netpet, 2025-08-07:
Fixed:

I use a stripped Mennekes Cable. This does not have a pilot cable inside, so no connection between the stripped cable PP-Pin and the SmartEVSE. Instead they have two 220Ohm Resistor connections directly in the plug head. So if you measure even the cutted cable you get 220Ohm between PE and PP in both side of the plugs. 

Because of the missing PP-Cable I had to put a direct connection of 220Ohm between the PP and PE at the SmartEVSE side.

I have reset the firmware to 3.8.5. now it seems to work again (no reset of the settings yet). I will try the debug with both versions (3.8.5 and 3.9.1) later when the sun stopped shining ;-) and have a look again tomorrow, as the problem always occurs when the charging was stopped a longer time. 

But thanks for your great support. As I see, the source code is in python, so I will have a try to implement the CP logging later.

netpet, 2025-08-09:
So on 3.8.5 it started, still the problem when Tesla stops with the error message on Tesla side, but at least it restartet without unplugging. 

The Switch doesnt seem to do anything. Also direct connecting two wires between SW and GND doesnt show any effect. No Toggling, no respone at all.

Controlling the SmartEVSE with MQTT works fine. If interested how it looks: charging.innvillage.at

dingo35, 2025-08-09:
If I short circuit pins 4 (GND) and 5 (Switch) the LCD screen goes to "Access denied" and back, so I cant reproduce your problem.

Are you sure your low-power connector is seated correctly?

schup011, 2025-09-06:
May I just ask a related question: I also have the problem with my VW ID.7 that "sometimes" in solar mode, after SmartEVSE has interrupted charging, it won't restart again. Also in this case, the vehicle reports "fail" (I have carconnectivity-mqtt running and can see the charging status) and usually (not always) you need to plug out and in again to restart charging. It does not happen every time, and I cannot really correlate conditions with it. But if it occurs, it is after having charged and stopped by SmartEVSE. 

Even if I also have read that VW ID models seem to have a problem with car sleep in these cases, as I am reading here that Teslas show similar problems, I am asking whether there is some kind of SmartVESE behavior that could provoke such an error in the vehicle electronics:

- Maybe the order the power is disconnected (in solar mode, I am charging 1 phase)? 
- Or the PWM signal during shutdown?

mstegen, 2026-01-11:
@netpet Did you solve the problem, or is it still present?