SmartEVSE/SmartEVSE-3 #106: Immediate stop of charging in solar mode - timer not working

schup011, 2025-09-05:
**Describe the bug**
When in solar mode and mains current goes beyond adjusted level, charging stops immediately, I only see the stop timer (upper right of the display) a few seconds, then it stops.

**Upload your config**
Press the "raw" button in the SmartEVSE webserver and upload the json settings file, so we can look at your configuration.

**Upload a debug log**

**To Reproduce**
Steps to reproduce the behavior:
Solar mode. Mains data delivered by MQTT ("API"). 
Start current: 1A
Max import: 2A
Stop time: 1 minute
Mains import has to go over adjusted max import (2A).

**Expected behavior**
After mains import is more than 2A, timer should start and then stop charging after expired. If mains import gets less than 2A during timer operation, timer should cancel.

**Screenshots**

mstegen, 2025-09-05:
Hi, 
It has something to do with the CAPACITY setting. If i set this to 60A, i see the same behaviour.
Are you in Belgium? Otherwise i would recommend to set it to Disabled.

schup011, 2025-09-05:
OK, I did not know this. I do not really need the capacity setting, I just thought it was sufficient to set it high enough.

I adjusted the settings according to your recommendation and can confirm that it works. Thank you!

Still I would say this is a bug - capacity was not exceeded in my case.